hamish2014/FreeCAD_assembly2 #83: Assembly2 solver problems with 64bit Freecad under windows 64bit

rasomei, 2015-12-22:
Hi,
I have some older assemblys made with assembly2 modul under freecad 32bit windows 32bit.
This worked very fine. After changing to freecad 64bit and windows 64 bit there are problems with solving some constraints. Some parts are changing their position.
 When using freecad 32bit under windows 64bit everything is well.
Using mint 17.3 64 bit everything is also allright.

I used:
OS: Windows 8
Word size of OS: 64-bit
Word size of FreeCAD: 32-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

Assembly is ok.
____________________________________
OS: Windows 8
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6076 (Git)
Build type: Release
Branch: master
Hash: 0b0d258afd4ef8c9b42d03d5e43291d7214eeed6
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

Assembly is not ok, some parts changed their position
______________________________________________
OS: Linux Mint 17.3 Rosa
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6078 (Git)
Build type: None
Branch: master
Hash: eb5ec62ccb7bfc873abe08efdb0759e88cdd5c43
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

Assembly is ok.

Could this be a freecad or windows problem?

kind gegards

Ralf

[Baugruppe_K8200_Rahmen.zip](https://github.com/hamish2014/FreeCAD_assembly2/files/69351/Baugruppe_K8200_Rahmen.zip)

sgrogan, 2015-12-22:
Ralf,
Can you try with Win 0.15.4671 x64?

rasomei, 2015-12-22:
Hi,
thank you for your quick response.
But the problem is the same as with Freecad 0.16.6078. It seems, the problem could be windows or the linked windows libraries. 

OS: Windows 8
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

best regards

Ralf

hamish2014, 2015-12-23:
Hi Ralf,

Sounds frustrating.

The Assembly 2 workbench uses the direction vectors taken from an object.Shape attribute
It seems that these direction vectors are constructed differently (directionally inverted) depending the FreeCAD and OS combination.
The behavoir can be caused by almost any library.

Therefore i am afraid i cannot be of much help

rasomei, 2015-12-23:
Hi Hamish2014
in my opinion the problem is caused by any library, too. I will ask for a solution in the freecad forum.
It seems, I am the only one who use freecad on Windows 64 bit. Perhaps I have to switch to linux mint.

FYI:
saving a assembly, which was opened under Win 64bit (with incorrect position of some parts) and reopen in under Win 32 bit and pressing the solving button, the parts switch to  their correct position.
It's funny.
Perhaps there is someone in the community who has  the same problem.

best regards

Ralf

hamish2014, 2015-12-23:
I thought of a possible workaround for the issue of the constraint directions getting muddled up.
Assuming that the objects are already assembled in the correct orientation, change all the direction attributes to unassigned, and then solve the system.
Solving the system will then assign 'aligned' or 'opposed' according to the objects current orientation.

To try this workaround,
1) switch to the assembly file,
2) run the following in the python console

```python
for obj in FreeCAD.ActiveDocument.Objects: 
    if 'ConstraintInfo' in obj.Content:
        if hasattr(obj,'directionConstraint'):
            obj.directionConstraint = ["none","aligned","opposed"]
            obj.directionConstraint = "none"

            
import assembly2solver
assembly2solver.solveConstraints(FreeCAD.ActiveDocument)

```

Does this work for you?

This *repair* script will need to run every time the directionality changes (i.e. when you open the assembly2 on a system with different shape normal vectors, compared to the previous)

rasomei, 2015-12-23:
Hi Hamish2014,

there is no effect to my assemblys. Not all parts are changing their position or their location.
It seems, it is depending on wha

rasomei, 2015-12-23:
Hi Hamish 2014,

the script will not take any effect to my assemblies. Not every part is changing it's location.  I think, it is depending on what kind of constraint I used. In most cases the angle constraint or an plan to vertex constraint is mismatching. I wonder, if the numbering of faces is different between 32bit and 64 bit.
Could you open my assembly attached to my first mail (Baugruppe_K8200_Rahmen.zip) on win 64bit.
Then you will see, what my concern is.

Thank you in advance

Ralf

easyw, 2017-01-13:
as in this FC fprum discussion:
http://forum.freecadweb.org/viewtopic.php?f=3&t=13619&start=40#p153189
the problem has been proven to be related to the windows version of **numpy** both for FC 32b and 64b
Without substituting the numpy package, a partial workaround is to change arctan2 with atan2 of math... unfortunately this doesn't cover all the solver problem in windows...
Maurice