hamish2014/FreeCAD_assembly2 #85: [suggestions] : rotation capability in addition to translation one + hide selected parts

paul18fr, 2015-12-31:
Hi,

*a) rotation capability*
The 2 latest icons are really helpfull (flip direction and lock/unlock rotation DOF); in the same idea of facilitating the use of the module, it might be interesting to easily **rotate** the part/assembly on which we are working on (especially when we've dozens of parts to manage)

currently we can only do it the combo view "by hand".

*b) hide specific part by (right) clicking on*
In my current model, I've dozens of parts and I create as many sub-assemblies as I can in order to lighten the solver calculations, so we I went to **hide** a part, the complete sub-assembly is hidden and sometimes it's not convenient

let me saying it again, Assembly2 is really a **great module** done by a great guy.

Happy new year

Paul

hamish2014, 2016-01-01:
Thanks Paul,

Interesting suggestion.

Please note however the next item on my Assembly2 todo list is with regard to accommodating the numerical (float) errors which sometimes cause the solver the crash.
Hopefully i will get time over the coming months to address this issue.

Wishing you great new year too.

paul18fr, 2016-01-01:
Hi Hamidh2014

I can figure in how busy you are, and it's is a suggestion coming from "practical" concerns (see the picture - still under development - original project at https://github.com/carlosgs/Cyclone-PCB-Factory/releases); it'll be an easier manner to proceed while using the comboo view.

In the meantime, I'll continue to use this module for my projects and the CAD training courses I do in our Fablab.

Release 2.2
[image]

Thanks for your works

Paul