hamish2014/FreeCAD_assembly2 #87: Problème avec version 0.16

FMD14, 2016-01-02:
Bonsoir,

Une dll ne charge pas au lancement FreeCad 0.16
[image]
Que faire ?

FMD14

jbet, 2016-01-07:
Vous n'êtes pas sur le bon forum, ici c'est assembly2, et non Freecad.
Il y a un forum français pour Freecad. 
Passer plutôt par là, et cela sera plus efficace.
Préciser bien l'OS que vous utilisez...

En espérant avoir répondu à ta question.

FMD14, 2016-01-07:
Bonjour,

 

Jai posté une question au assembly2 sur le forum Français mais personne
peut me dépanner

 

Je suis sous vista et la version 0.16 

 

Par contre sous 0.15 ça marche

 

 

Cordialement

 

 

 

 

 

  _____  

 

<https://github.com/hamish2014/FreeCAD_assembly2/issues/87#issuecomment-1696
76665>  it on GitHub.
<https://github.com/notifications/beacon/APwOvnmQWdNp-U2AoZ75pRllnKkST5Koks5
pXmr_gaJpZM4G9kpk.gif> 

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

jbet, 2016-01-07:
La version 0.16, est disponible, mais elle est peut avoir des bugs (enfin je n'en ai pas encore vue)
Je suis sous Ubuntu Trusty, et Freecad et la version .16 fonctionne avec Freecad.

Ce qui vous arrive est quand même étonnant, car Freecad et Assembly 2 sont deux logiciels différents, et Assembly 2 récupère les fichiers Freecad.
Donc s'il marche sur la version 0.15, il devrait marcher sous la version 0.16

Ce que ferais a votre place, en attendant que que la version .16 sous officiel, c'est de rester sur la 0.15, en attendant des nouvelles, coté Hamish.

Autre idée : Passer votre poste sous Ubuntu ( 2GO de mémoire vive sont le bienvenu), en double boot, puis vous choisirez au démarrage, Ubuntu ou Vista

Bon courage.

----- Mail original -----

<https://github.com/hamish2014/FreeCAD_assembly2/issues/87#issuecomment-1696 
<https://github.com/notifications/beacon/APwOvnmQWdNp-U2AoZ75pRllnKkST5Koks5 

https://www.avast.com/antivirus