hamish2014/FreeCAD_assembly2 #89: AxialConstraint with 3d curve

FlachyJoe, 2016-01-05:
Hi,
Here is an idea to get an axis from a 3d curve as an helix :
To be added in getSubElementAxis
```
        portion=100
        axis=FreeCAD.Vector(0,0,0)
        for t in range(portion):
            t2=edge.getParameterByLength(edge.Length)*t/portion
            axis=axis.add(edge.Curve.tangent(t2)[0].cross(edge.centerOfCurvatureAt(t2).add(edge.valueAt(t2).negative())))
        axis.multiply(1/axis.Length)
        return axis
```
*edge.centerOfCurvatureAt(0)* can maybe used in getSubElementPos

I tried to implement this but got *quadraticLineSearch: norm search direction is 0, aborting!*

To be continued...

FlachyJoe, 2016-01-06:
So, it's working but as the axis is not the true one but an approximation I can't use this in combination with a planar constraint to fix the last 2 DOF. (What I think...)
Can somebody help to get a sufficient approximation ?
This is my patch :
At the end of getSubElementPos
```
    if pos <> None:
        return numpy.array(pos)
    else:
        #3d curve
        pos=edge.centerOfCurvatureAt(0)
        return numpy.array(pos)
```

At the end of getSubElementAxis
```
    if axis <> None:
        return numpy.array(axis)
    else:
        #3d curve
        portion=128
        axis=FreeCAD.Vector(0,0,0)
        factor=edge.getParameterByLength(edge.Length)/portion
        for t in range(portion):
            t2=factor*t
            axis=axis.add(edge.Curve.tangent(t2)[0].cross(edge.centerOfCurvatureAt(t2).add(edge.valueAt(t2).negative())))
        axis.multiply(1/axis.Length)
        return numpy.array(axis)
```

Added in assembly2lib.py
```
def edgeSelected( selection ):
    if len( selection.SubElementNames ) == 1:
        subElement = selection.SubElementNames[0]
        if subElement.startswith('Edge'):
            return True
    return False
```

Modified ValidSelection in AxialConstraint.py
```
def ValidSelection(selectionExObj):
     return cylindricalPlaneSelected(selectionExObj)\
         or LinearEdgeSelected(selectionExObj)\
         or edgeSelected(selectionExObj)
```

hamish2014, 2016-01-07:
Nice work Flancy Joe!

Please note that your enhancements wont be implemented in this branch for a while, since
I am taking the rest of the month of from FreeCAD_assembly2 and FreeCAD_drawing_dimensioning...

FlachyJoe, 2016-01-10:
The position is now the barycenter of centers of curvature.
Work well...
```
--- FreeCAD_assembly2/assembly2lib.py	2016-01-10 22:21:06.615379958 +0100
+++ My_assembly2/assembly2lib.py	2016-01-10 21:31:10.114752737 +0100

@@ -297,9 +297,16 @@
                 return error_normalized < 10**-6
     return False
 
+def edgeSelected( selection ):
+    if len( selection.SubElementNames ) == 1:
+        subElement = selection.SubElementNames[0]
+        if subElement.startswith('Edge'):
+            return True
+    return False
+
 def getObjectEdgeFromName( obj, name ):
     assert name.startswith('Edge')
     ind = int( name[4:]) -1
     return obj.Shape.Edges[ind]
 
 def CircularEdgeSelected( selection ):

@@ -399,20 +406,28 @@
                 centers = numpy.array([a.Center for a in arcs])
                 sigma = numpy.std( centers, axis=0 )
                 if max(sigma) < 10**-6: #then circular curce
-                    pos = centers[0]
+                    return centers[0]
             if all(isinstance(a, Part.Line) for a in arcs):
                 lines = arcs
-                D = numpy.array([L.tangent(0)[0] for L in lines]) #D(irections)
-                if numpy.std( D, axis=0 ).max() < 10**-9: #then linear curve
-                    return lines[0].value(0)
+                if len(arcs)>1 :
+                    D = numpy.array([L.tangent(0)[0] for L in lines]) #D(irections)
+                    if numpy.std( D, axis=0 ).max() < 10**-9: #then linear curve
+                        return lines[0].value(0)
     elif subElementName.startswith('Vertex'):
         return  getObjectVertexFromName(obj, subElementName).Point
     if pos <> None:
         return numpy.array(pos)
     else:
-        raise NotImplementedError,"getSubElementPos Failed! Locals:\n%s" % formatDictionary(locals(),' '*4)
+        #3d curve
+        portion=128
+        mult=edge.getParameterByLength(edge.Length)/portion
+        pos=FreeCAD.Vector(0,0,0)
+        for t in range(portion):
+            pos=pos.add(edge.centerOfCurvatureAt(mult*t))
+        pos=pos.multiply(1./portion)
+        return numpy.array(pos)
 
 def getSubElementAxis(obj, subElementName):
     axis = None
     if subElementName.startswith('Face'):
@@ -453,6 +468,12 @@
     if axis <> None:
         return numpy.array(axis)
     else:
-        raise NotImplementedError,"getSubElementAxis Failed! Locals:\n%s" % formatDictionary(locals(),' '*4)
-
-
+        #3d curve
+        portion=128
+        mult=edge.getParameterByLength(edge.Length)/portion
+        axis=FreeCAD.Vector(0,0,0)
+        for t in range(portion):
+            t2=mult*t
+            axis=axis.add(edge.Curve.tangent(t2)[0].cross(edge.centerOfCurvatureAt(t2).add(edge.valueAt(t2).negative())))
+        axis.multiply(1/axis.Length)
+        return numpy.array(axis)

```

```
--- FreeCAD_assembly2/axialConstraint.py	2016-01-10 22:21:06.615379958 +0100
+++ My_assembly2/axialConstraint.py	2016-01-05 16:00:27.123788000 +0100
@@ -9,7 +9,8 @@
 
 def ValidSelection(selectionExObj):
      return cylindricalPlaneSelected(selectionExObj)\
-         or LinearEdgeSelected(selectionExObj)
+         or LinearEdgeSelected(selectionExObj)\
+         or edgeSelected(selectionExObj)

```