hamish2014/FreeCAD_assembly2 #91: instructions for installing on Macintosh

piffpoof, 2016-01-10:
here are some instructions for installing the Macintosh, if you want to append them to the readme.md

Macintosh Installation Instructions
---------------------------------

  * download the git repository as ZIP
  * assuming FreeCAD is installed in "/Applications/FreeCAD/v 0.15", go to "/Applications/FreeCAD/v 0.15" in the Browser, and select FreeCAD.app
  * right-click and select "Show Package Contents", a new window will appear with a folder named "Contents"
  * single-click on the folder "Contents" and select the folder "Mod"
  * create new directory named "assembly2"
  * unzip downloaded repository in "C:\PortableApps\FreeCAD 0_15\Mod\assembly2"

piffpoof, 2016-01-10:
My error, the last 2 lines should read as follows:

  * in the folder "Mod" create a new folder named "assembly2"
  * unzip downloaded repository in the folder "Contents/Mod/assembly2"[/code]

tiborsimon, 2016-03-10:
I can confirm this on OS X El Capitan 10.11.3, MacBook Pro (Retina, 13-inch, Late 2013), 2,4 GHz Intel Core i5.

OS: Mac OS X
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.9
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

BPLRFE, 2016-11-17:
Integrated in readme.md
Thanks Dean (piffpoof) for testing and reporting ...

closed 2016-11-17 by BPLRFE