hamish2014/FreeCAD_assembly2 #92: FreeCAD crashes when locking rotation DOF

paul18fr, 2016-01-11:
Hi,

While it worked quie well so far, now freecad crashes when I'im trying to lock rotation DOF's.

In the attachement (<b>the original file is a .7z one renamed into pdf to be accepted by the system</b>), I'm noticing it for AxialContraint 3 & 4.

I'm under Ubuntu, using the latest daily Freecad build (i.e. 0.16xxxx).
[spindle.pdf](https://github.com/hamish2014/FreeCAD_assembly2/files/86379/spindle.pdf)

I'm sceptic: am I doing something wrong? I've been experimenting troubles since a few time ago

hope it helps

Paul

f3nix, 2016-01-12:
Hi,
I can confirm the crash. Tried with axial constraints and managed to crash FreeCAD in few different ways...

The only approach that worked for me (to get 0 DOF) was using circular constraints.
I don't know why, but my assembly is 0,5mm taller than yours (185,5 vs 185,0).

[image]

7z archive with new assembly.
[spindle_new.pdf](https://github.com/hamish2014/FreeCAD_assembly2/files/87875/spindle_new.pdf)

Cheers,
Mateusz

paul18fr, 2016-01-13:
Thanks Mateusz for your answer,

The main goal of the post is to share a feedback on troubles and workaround can be found as you’re suggesting.

I feel changes in the module since few time ago, especially when I see what I was able to do before such troubles appear (see attachment – <b>as usual it is a 7z file</b>); the project includes dozens of parts and sub-assemblies.

Now I’ve an issue when after adding a new part in a sub-assembly, I tried to update the master assembly.

In a previous post, Hamish2014 told he’s currently improving the assembly solver, so I’ve to champ at a bit :-)

Paul

[PCB_cyclone_factory.pdf](https://github.com/hamish2014/FreeCAD_assembly2/files/88511/PCB_cyclone_factory.pdf)