hamish2014/FreeCAD_assembly2 #93: Error on Import button

web4deb, 2016-01-12:
Hi, when I click on the Import Part button, I get the following error:

Running the Python command 'importPart' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\assembly2\importPart.py", line 121, in Activated
    view = FreeCADGui.activeDocument().activeView()

'NoneType' object has no attribute 'activeView'

This is a new install on this computer v0.15, Nothing else has been installed except this assembly2 module.

Thanks!

f3nix, 2016-01-12:
Hi,
you have to create a new, empty document first.

File->New

And then you can start importing your parts.

HTH.

Cheers,
Mateusz

web4deb, 2016-01-12:
Thanks.  That works.  Sorry, I'm still a noob at using this.  ;-)

closed 2016-01-12 by web4deb