hamish2014/FreeCAD_assembly2 #97: Recommendations for Using

Rapid-Fab, 2016-02-17:
Hi Hamish, Will your workflow be included in the upcoming FreeCad updates or are there others working on this module?  Secondly, do you have any hints for using this workflow...  I would also be interested in assisting with creating manuals if that is of interest. ..  Now to my issue...  I have been trying to keep assembly simple, but it keeps deciding to drop the constraints, and go its own way. Specifically what happens when a part is (usually accidentally) repositioned with the move part command after restraint(s) assigned.  Eric

Rapid-Fab, 2016-02-17:
oh, and sometimes the parts all just seem to disappear. Not sure what is with that.

hamish2014, 2016-07-16:
Thank you for your post.
Please note that I am disabling the Issue tracker for assembly2.
I am doing so because over the last year I have not had the time to respond to any assembly2 issue properly (and wont be able to do in the future either).
I hope that you have managed to resolve your issue in the meantime.