hamish2014/FreeCAD_assembly2 #98: Ubuntu. All Constraints Fail

addis2123, 2016-02-19:
While trying to create a constraint in FreeCAD using assembly2 I receive an error every time.

Running the Python command 'addPlaneConstraint' failed:
Traceback (most recent call last):
  File "/home/Addis2123/.FreeCAD/Mod/FreeCAD_assembly2/planeConstraint.py", line 87, in Activated
    secondSelectionGate = PlaneSelectionGate2() )
  File "/home/Addis2123/.FreeCAD/Mod/FreeCAD_assembly2/assembly2lib.py", line 49, in __init__
    FreeCADGui.Selection.addSelectionGate( selectionGate )

must be string, not instance

terrylm, 2016-03-02:
I am also running Ubuntu, with no problems now.  At first it was crashing, but it turned out I had some how freshly installed an old version of the workbench.  I am also runing FreeCAD 0.16.

Maybe make sure you are using 0.16, and reinstall the workbench via the auto-workbench-installer-macro.

addis2123, 2016-03-06:
I saw this seven days to late. The computer has fried the hard drive. Will tell you if the new install works

addis2123, 2016-03-07:
It seems to be working now that only freecad and Assembly 2 are the only thing installed.

hamish2014, 2016-07-16:
Thank you for your post.
Please note that I am disabling the Issue tracker for assembly2.
I am doing so because over the last year I have not had the time to respond to any assembly2 issue properly (and wont be able to do in the future either).
I hope that you have managed to resolve your issue in the meantime.