hamish2014/FreeCAD_assembly2 #100: Solver does not always take advantage of existing degrees of freedom.

terrylm, 2016-03-04:
Example:

An engine model.

The piston is axial constrained to the cylinder. The piston hangs in space out the top of the cylinder.

The rod is circular edge constrained to the crank. The rod is not pointed in the proper direction.

An attempt is made to place either (not both) an axial constraint or a circular edge constraint to connect the rod to the piston.  It fails. Apparently is does not attempt to slide the piston inward and rotate the rod on the crank such that they can come together.

If only one part has a need to move an existing degree of freedom, then it will succeed, but when both parts being joined have to each move on a degree of freedom, it fails. At least that is what looks to be the problem.

If there was a way to manually maneuver the parts, one could work around this problem?

Thanks for all the great work on Assemble2.

terrylm, 2016-03-07:
Working from the animation function as an example (copy and hack), I built a manual DOF editor.

It works mostly, save for one little detail, once it has the new value of a particular DOF, I do not know where to put it so that It becomes a change in the document?

Any help appreciated.  Is there a better place to discuss this?

Thanks

hamish2014, 2016-07-16:
Thank you for your post.
Please note that I am disabling the Issue tracker for assembly2.
I am doing so because over the last year I have not had the time to respond to any assembly2 issue properly (and wont be able to do in the future either).
I hope that you have managed to resolve your issue in the meantime.