hamish2014/FreeCAD_assembly2 #102: New constrains (symetric and kinematic)

efyx83, 2016-03-16:
Hi, it would be perfect to add symmetric constrain. It should look like, first you show 2 planes of first object, then two planes of second object and it should positioning it symmetrically.
Kinematic constrain (maybe some kind of cursor - instead of animation) to check whole range of movement. It should rebuild in real time. For example you have cylinder in drilled hole and you can move it by dragging up and down to check how it works (in this case up-down).

terrylm, 2016-03-18:
If it helps you with that last point, try my fork, it has a dialog to manually change any DOF (Degree of Freedom).  What you are talking about is a degree of freedom.

https://github.com/terrylm/FreeCAD_tinkering

efyx83, 2016-03-21:
Each time I need to compile, new version of freecad when I need some additional module? I meant I've compiled last time version with Techdraw, do I need to clone this one and compile as other instance of freecad? Is there any way to join it to my version of freecad as a module or macro?

hamish2014, 2016-07-16:
Thank you for your post.
Please note that I am disabling the Issue tracker for assembly2.
I am doing so because over the last year I have not had the time to respond to any assembly2 issue properly (and wont be able to do in the future either).
I hope that you have managed to resolve your issue in the meantime.