hamish2014/FreeCAD_assembly2 #105: conditional deactivation of constraints

microelly2, 2016-05-06:
Hi Hamish,

I want to improve the integration of assembly2 and animation wb.

First I want to activate/deactivate  constraints on conditions: The simplest  case is to use one flag:if the viewobject is hidden then the assembly constraint is not processed and vice versa.
The more sophisticated case is to add a property to the constraints objects and configure the solver that it solves a constraint only it this property has a special value. This allows for example to have constraint groups activated in a gearing.
This allows once a group is assembled to deactivate the constraints for this group and preserve the calculated placements fo the group  for the processing of further "higher" constraints.

The third  step is to have restrictions on the degrees of freedom. I want to have upper+lower bound pairs so the system can solve both extrema and I can animate a move betweeen this positions.
I think this all can be done compatible to your work.

Do you see some problems to go this way?

microelly2, 2016-05-09:
I have started to get more benefit from assembly to animation
https://www.youtube.com/watch?v=v9pmMegpxN8
I will play a little bit to learn the internals of the assembly wb, and then find a way to cooperate between workbenches
It is a fantastic advantage: put parts together to build the machine, calculate some key positionings with assembly and then interpolate the work pathes to get the machine running inside it's geometry.

hamish2014, 2016-05-11:
Video looks great microelly2.
Agreed it a fantastic advantage to visualize a machine/mechanisms to motion, to aid the design process.

assembly2's animate degrees-of-freedom was not designed for this however.
animate dofs was designed so that
- i could visualize the solver's dof reduction, and 
- so that users can see which parts are not fully constrained.

So if your workbench can animate a machine motion that is fantastic, I am sure the community will find it very useful.