hamish2014/FreeCAD_assembly2 #110: Why this is not already merged into freecad ?

rwoloszyn, 2016-06-24:
This tool is awesome. Did you wrote to Freecad to merge your code into FreeCad main repo ?

hamish2014, 2016-07-01:
Hi see #115

hamish2014, 2016-07-01:
Hi  @rwoloszyn,

To my knowledge there are no plans to merge assembly 2 into FreeCAD v0.16 or v0.17.
This statement means very little though, because I have not being keeping up with developments in FreeCAD.
Last I heard the FreeCAD team was busy with assemblyNext.
@yorikvanhavre?

If it turns out that assemblyNext is far from completed, I will contact the FreeCAD team about what needs to be done to merge assembly2 in FreeCAD v0.17

Cheers,

rwoloszyn, 2016-07-03:
@hamish2014. This tool is awesome and it will be nice to have it in free cad official version. 
Anyway thanks for answer.

yorikvanhavre, 2016-07-03:
Basically there is no problem in merging assembly2 into FreeCAD itself, @hamish2014 if you wish to do so, we'll do it. You decide. However, I am not sure it is really better, for a series of reasons:

- It would somehow "require" you to become more involved into the FreeCAD forum, because many questions and issues will begin to appear there, you would need to stick more to the new versions, etc. Here, you can work more as you wish, at your own convenience.
- You would have to make pull requests everytime you need to change something, and keep the whole FreeCAD code in sync. Which is a hassle, and much more annoying to maintain than in its current form. So there is a risk: more work for you, therefore you'll be able to do less coding.
- The google summer of code project currently going on aims at implementing a decent plugins manager for FreeCAD. Once that is done (and it will be, even if the GSOC project doesn't finish we will finish it), installing external workbenches such as this one will be a piece of cake. 
- I will also ensure that workbenches authors have their names in the credits list, since they are fist-class FreeCAD developers too. I'm working on that already.

On the other hand, having workbenches merged into the FreeCAD master code could help for example, to keep unmaintained workbenches alive (But this is not the case of assembly2). Also doing so doesn't make magically appear other developers to  work on it. The tendency is actually the contrary, see for ex. the openSCAD or Ship workbenches, only their original authors ever worked on them.

And finally indeed, now that PartDesignNext is merged, people there are starting to work on AssemblyNext! 

So I would advocate for keeping assembly2 here for the time being. But, if you feel it's better for this or your other workbench to be in FreeCAD master, just tell me, I'll gladly do it.

hamish2014, 2016-07-05:
Thanks for your detailed response @yorikvanhavre 
2 questions, 

1) both assembly2 and drawing_dimensioning require numpy, would this a problem for the merge?
2) is the FreeCAD team currently working on drawing dimensioning functionality? If not, then I will try to merge drawing dimensioning with FreeCAD 0.17 first (based on that experience, I will see about assembly2)

If these workbenches are merged into FreeCAD or not,
I am planning disable the issues feature on both drawing_dimensioning and assembly2.
The truth is I am not able to attend to these issues any more, and the repo's should be updated accordingly.
i.e. the repos should not have Issues enabled if not one is going to look at those issues...

yorikvanhavre, 2016-07-05:
1) This should be decided with the people who package for Windows (sgrogan mainly). I *think* numpy is already in that package, but not sure. On Linux there is no problem, and on Mac I don't know, that should be investigated too (I think sgrogan will know too).
2) yes. There is wandererfan's TechDraw branch that should be merged in the next days. This should have many tools that drawing_dimensioning has. The idea is that TechDraw ends up replacing Drawing (for now both will be kept separate)

I think indeed progressively what assembly2 and drawing_dimensioning do will be implemented directly in C++ in FreeCAD (AssemblyNext and TechDraw). So their use should become less important. But, the same happened with Draft and Sketcher (many draft ideas have been used in the sketcher), but Draft is still useful today, and I think the same will happen with drawing_dimensioning and assembly2.

If you don't have more time to maintain these two workbenches, indeed it would be a good reason to merge them in FreeCAD. I would suggest you make a post about that on the FreeCAD forum, it will be easier to discuss packaging issues there with other people involved...

sgrogan, 2016-07-15:
@hamish2014 @yorikvanhavre 

It's no problem supporting numpy. It's already in the Win bundle, it's available in homebrew which is what we are using for osx, and we can add it to the depends for the PPA.

I think it would be a shame to see this work die because you don't have the time to maintain it. There have been quite a few new people asking for development stuff to work on, maybe you could recruit a successor on the forum. I know from the traffic on the forum that these workbenches both have a large user base.

hamish2014, 2016-07-16:
@sgrogan @yorikvanhavre 

Thank you both for your input.
I am going to create 2 posts on the FreeCAD forum and we can see what happens from there.

hamish2014, 2016-07-21:
http://forum.freecadweb.org/viewtopic.php?f=17&t=16591
http://forum.freecadweb.org/viewtopic.php?f=17&t=16590