hamish2014/FreeCAD_assembly2 #115: are there any plans to port this to v0.17

mgiaco, 2016-06-30:


hamish2014, 2016-07-01:
Hi mgiaco,

FreeCAD_assembly2 should work in FreeCAD v0.16.
There are no plans at the moment to port to FreeCAD v0.17

Cheers,

hamish2014, 2016-07-21:
http://forum.freecadweb.org/viewtopic.php?f=17&t=16591

easyw, 2016-11-15:
I started to go into FC 0.17 new Part Design and I found a way to let Assembly2 WB to load correctly a Part made with a Visible Body and with sub Parts
http://forum.freecadweb.org/viewtopic.php?f=9&t=17742#p144714

there is a very small diff in the code of the file "importPart.py" of the WB @ line 43

```
   visibleObjects = [ obj for obj in doc.Objects
                       if hasattr(obj,'ViewObject') and obj.ViewObject.isVisible()
                       and hasattr(obj,'Shape') and len(obj.Shape.Faces) > 0 and 'Body' not in obj.Name] # len(obj.Shape.Faces) > 0 to avoid sketches

```
I added only
`and 'Body' not in obj.Name
`

Maurice

BPLRFE, 2016-11-17:
After some testing i can confirm that this works with FreeCAD 0.16.6706 on Windows 7 - 64 bit (able to import a file saved with version 0.17)
Also tested an confirmed with FreeCAD 0.17.8922 on Windows 7 64-bit.
Code merged into master thanks for contributing.

closed 2016-11-17 by BPLRFE