hamish2014/FreeCAD_assembly2 #116: Workbench did not appear after Installation on freecad v0.16 Windows

wlchui, 2016-07-03:
i tried installing according to the steps by unzipping the zip into the mod directory of my Freecad directory.  The workbench never appeared even after a restart on freecad.

I am using Windows XP 32 bit version and freecad version 0.16.

rsxy, 2016-07-05:
Hi, I am having the same problem under Win 7 (32 bit), also with the DrawingDimensioning workbench.
However, both add-ons do appear using the 0.17 dev version.

hamish2014, 2016-07-16:
Thank you for your post.
Please note that I am disabling the Issue tracker for assembly2.
I am doing so because over the last year I have not had the time to respond to any assembly2 issue properly (and wont be able to do in the future either).
I hope that you have managed to resolve your issue in the meantime.

normandc, 2017-09-16:
FYI, this is not a problem with Assembly2, but with the 32- bit version of FreeCAD 0.16 on Windows. It is apparently due to an incompatible version of numpy, a python component installed with FreeCAD.

See the FreeCAD documentation for a fix: https://www.freecadweb.org/wiki/How_to_install_additional_workbenches#General_troubleshooting

Therefore I think this issue should be closed.

BPLRFE, 2017-09-17:
Thanks for mentioning that, Normand. I am closing the issue and will add a remark in the assembly2-wiki
here on github.
https://github.com/hamish2014/FreeCAD_assembly2/wiki

closed 2017-09-17 by BPLRFE