hamish2014/FreeCAD_assembly2 #119: Unsolvable Non-Conflicting Constraints

Lotharyx, 2016-07-06:
I'm attaching a .fcstd file with an example.  I am trying to assemble parts from 80/20 (metal extrusions & hardware).  There is a linear bearing that consists of an "h"-shaped bracket and bearing pads.  The pads bolt to the bracket with two bolts each.

I should be able to create axial (or circular edge) constraints on the bolt holes and an opposing plane constraint between the bearing pad and the bracket, but the solver fails when I add the plane constraint.  

In the attached file, I have already created the axial constraints for the bolt holes.  To reproduce the issue, try to create a plane constraint to mate the large flat face of the bearing pad with the matching interior face of the bracket.  I have attached a second file with that constraint applied (and unsolvable).
[without_plane_constraint.zip](https://github.com/hamish2014/FreeCAD_assembly2/files/350505/without_plane_constraint.zip)
[with_plane_constraint.zip](https://github.com/hamish2014/FreeCAD_assembly2/files/350506/with_plane_constraint.zip)

This issue prevents me from using assembly2 workbench successfully.  This may be the same issue as #111, #112, or #113.  I have not tried downloading the presentation files from them to find out.

hamish2014, 2016-07-16:
Thank you for your post.
Please note that I am disabling the Issue tracker for assembly2.
I am doing so because over the last year I have not had the time to respond to any assembly2 issue properly (and wont be able to do in the future either).
I hope that you have managed to resolve your issue in the meantime.

easyw, 2017-08-04:
this can be successfully closed with latest A2 and the numpy release 1.12.0 from here
https://forum.freecadweb.org/viewtopic.php?f=3&t=13619&start=50#p159786
[assembly2 constraint problem fully constrained.zip](https://github.com/hamish2014/FreeCAD_assembly2/files/1201632/assembly2.constraint.problem.fully.constrained.zip)
tested on win 10, FC0.17 11731 and numpy 1.12.0

closed 2017-08-04 by BPLRFE