hamish2014/FreeCAD_assembly2 #120: Changing the licence to LGPL-2.1

hamish2014, 2016-07-19:
Hi @fcazorla @f3nix and @easyw 

When i created the FreeCAD_assembly2 I choose a licence which is incompatible with FreeCAD's licence.
Therefore I would like to change the licence to the GNU Lesser General Public License, version 2.1
This gives the FreeCAD team to option to use some of the code should they wish to.

As you are all contributors to assembly2 code belongs to all of us, so i need your permission to re-licence.

Thank you,

fcazorla, 2016-07-19:
Of course, you have my permission.

Thanks to you, for this useful code :)

easyw, 2016-07-19:
Hi @hamish2014 
you certainly have my permission...
thank you again for your **great WB!**
Maurice

f3nix, 2016-07-20:
Hello @hamish2014!

It is ok with me.

Thank you for your great work! :)
Very, very useful and saved me a ton of time!

Cheers,
Mateusz

hamish2014, 2016-07-21:
Thanks for the kind words everyone :D 
I will re-license now.

closed 2016-07-21 by hamish2014