hamish2014/FreeCAD_assembly2 #131: Align icons to FreeCAD guidelines

agryson, 2017-04-02:
I recently finished aligning the FreeCAD icons to a unified set of guidelines. [It's been suggested on the forum that i do the same for Assembly2](https://forum.freecadweb.org/viewtopic.php?f=34&t=21420&start=10)
I just want to clear this with the team taking care of this workbench. If you're happy to let me do it, just assign this ticket to me and I'll get to work, if not, simply close the ticket.

BPLRFE, 2017-04-02:
Hello Alexander.

Thanks for taking care of this. I think a "shape-up" of the icons would be a very good idea.

easyw, 2017-04-02:
Thx @agryson for taking care of it!
Maurice :smile:

agryson, 2017-04-02:
Great, expect a PR by end of week.

agryson, 2017-04-06:
For the curious amongst you:

Before:
[image]

After:
[image]

easyw, 2017-04-07:
Oooohh! 😸 
Thx so much! 
Maurice

agryson, 2017-04-07:
If you like, we don't do it anymore for PRs, github suffices, but if you prefer I can post a topic with a sprite sheet for review.

easyw, 2017-04-08:
@agryson I posted a comment at the forum before seeing your reply...
https://forum.freecadweb.org/viewtopic.php?f=9&t=17742&p=169102#p169102
I just noticed that the maintainers are more active at the forum than here :smile:

agryson, 2017-04-08:
No problem, thanks for doing it!

commit 1d5929bb (2017-04-08): Compile resources after artwork alignment, fixes #131
closed 2017-04-08 by BPLRFE