hamish2014/FreeCAD_assembly2 #133: Assembly errors upon parts update

ankou29666, 2017-04-25:
Hi everyone.

I have created an assembly from a few parts, but the problem I am facing is that after modifying a part, updating the assembly often results in errors, I mean plane constraints fancy refer to wrong faces with completely whimsical results.

Please correct me if I'm wrong, but it looks like the constraints makes for example Part003.Face045 match with Part011.Face123. It also looks like the faces are numbered by the order they appear in the Brep files. In other words, these numbering is not quite random, but not quite deterministic either. It looks like pseudo-random. 

The problem is if for example Part011 is modified and Face123 moves to 137th, the constraint remains upon Face123 instead of being updated to Face137.

Is there any trick to solve this kind of issue ?

the-snowwhite, 2017-04-25:
Hmmm .. base 12 ,?Excused also.

BPLRFE, 2017-04-25:
It may depend on the Operating system. If using 0.16.6706 64-bit you may get good results
manually replacing your NUMPY-Library like described here:
https://forum.freecadweb.org/viewtopic.php?p=159786#p159786
[It seems a buggy version of NUMPY was compiled with 0.16.6706 on Windows ...]
For german users i made a video showing the process:
https://www.youtube.com/watch?v=u7190e6hP1Q&index=25&list=PLsrwVwvqYb8EZyMnFb7Ogn4yjTZNLTKTB

ankou29666, 2017-04-25:
KUbuntu 16.10 FC 0.16.6707

ankou29666, 2017-04-25:
Ok, here's a little example to make it a bit more explicit : 
[test.zip](https://github.com/hamish2014/FreeCAD_assembly2/files/955940/test.zip)
Just unzip it, and open the assembly file and once you click the solver button you will see something like this : 
[image]
[image]

Now open the rising profile file, hide the part, edit the sketch, add a circle centered on the axes, and add a 2.1 mm radius constraint. Close the sketch, and toggle back the part to visible.

Get back to the assembly file, and update.
Here it results in this kind of bullshit : 
[image]
[image]

Constraint which originally bounded left_vertex_01.Face251 to rising_profile_01.Face1 and should not have been modified now bounds to rising_profile_01.Face11 instead.
Constraint which originally bounded left_vertex_01.Face134 to rising_profile_01.Face2 and should not have been modified now bounds to rising_profile_01.Face7 instead.
Constraint which originally bouned left_vertex_01.Face208 to rising_profile_01.Face45 has been correctly updated to rising_profile_01.Face46.

It's really amazing to see how such a minor modification in a part's geometry can break a complete assembly.

kbwbe, 2018-07-16:
Hello @ankou29666 ,
this behaviour is caused by FreeCAD itself as renumbering vertexes,edges,faces after very small model changes. As constraints of Assembly 2 are referring to these numbers, the constraints will point to false topology after reimport. There will be no change in Assembly 2 to avoid this as nearly impossible.

closed 2018-07-16 by kbwbe