hamish2014/FreeCAD_assembly2 #136: Extremely Slow Assembly Calculation

HexValid, 2017-06-13:
Hi,

When i use any function, calculation time takes very long times. (Sometimes longer than 3 minutes). I have a modern computer. This slowness only happens in assembly2 module. 
What is the reason of this?
Thanks.

> My Computer's Specs:
> Dual Core AMD CPU 4 core (3.2 GHZ)
> 4GB RAM
> Nvidia 5600 TI
> 
>

ankou29666, 2017-06-13:
The constraints solver is said by it's author to be really poor. This is known issue.
Another issue is due to Freecad files being zip archives. It takes some time to unzip the part's file and rezip it into the assembly file. Setting to no compression into the preferences will help, but won't make miracles.

HexValid, 2017-06-14:
Thanks for answer.
How can i disable compression?

kbwbe, 2018-07-01:
Biggest timeloss in Assembly2 occures if using importUpdateParts. Updating complex parts as for example the spindle motor from your screenshot will take a lot of time. I you do not want to wait some time, it is often better do delete the part from the assembly and insert it again. But you have to set your constraints again.

Some bugfixes will be done to assembly2 next future, but i think this behavior will not change.

closed 2018-07-01 by kbwbe