hamish2014/FreeCAD_assembly2 #138: Assembly2 Module does not generate Python Script for each icon clicked in the GUI

gitansh95, 2017-07-28:
Hi,

I want to see the Python code that is being executed when I click on an icon from the GUI, to be able to understand the code

Is there a way that  the code that is being executed for the each GUI icon could be printed on the FreeCAD Python Console, like other FreeCAD modules do?

Thanks.

ankou29666, 2017-11-26:
I've looked into the code and it seems to me pretty clear that the code has never been done to script around it.

kbwbe, 2018-07-01:
Assembly 2 is now maintained by me in a minimal way, to only keep it still running with fc-0.17 and fc-0.18
More i can not do. So this function will not be implemented. A look at the source files perhaps helps.

closed 2018-07-01 by kbwbe