hamish2014/FreeCAD_assembly2 #139: Add "Fix" Constraint

silopolis, 2017-08-08:
Hi, after a little time using he work bench a constraint that would "fix" an object placement and rotation would be useful in many cases (together with distance #22 and tangential #69 ones)
I my case, this would allow me to place a first object and set it's location "in stone", then add and position all other objects relatively using the other constraints.

BPLRFE, 2017-08-08:
This functionality is already there, but works only with parts imported into the assembly2-workbench (that is the intended original workflow ...).
> Create new document in FreeCAD
> switch to assembly2-workbench
> import part with first icon of assembly2-workbench
> select imported part in tree view
> look at its data tab
> the property of "fixed position" can be set to "true" or "false"
> if set to "true" the imported part is immobilized

easyw, 2017-08-08:
@BPLRFE 
may be that could be nice to be added to the wiki
https://github.com/hamish2014/FreeCAD_assembly2/wiki

silopolis, 2017-08-09:
@BPLRFE thank you soo much !

So modified this issue according to my experience to ask for a an additional icon/button to toggle this value without "digging" into the data tab and because it really is another constraint that should sit along its siblings.

Another helper would be a visual indicator showing that the object is fixed in space.  This could be a simple icon overlay in the tree view or a "fixConstraint..." under the object in the tree view. This would make the constraint visible as the others for consistency, and to avoid one to get mad trying for hours to have the object be moved by other constraints !...

Icon for this constraint could be an axis cross with the small green arrows like in others in the WB.

Now, only mistery left is why in the 20 or 30 imports I've done in the last days, only one object has had this "fixed position" value set and no others ?...