hamish2014/FreeCAD_assembly2 #153: Fix proposal for "assembly2/angleConstraint.py" and FC 0.17

ManuelKrause, 2018-04-01:
Sorry, people,
I'm relatively new to FreeCAD, am no programmer and don't know to use github properly for my purpose. I even don't know if this on here is the right place to report.

Recently I went from FC 0.16 to 0.17 to profit from it's advanced features. Unfortunately, the assembly2 addon failed to initialize, with an error message I can't recall.
Following the "recent" code changes by looooo, there is one missing thing left to make assembly2 work again:

in assembly2/angleConstraint.py
Line 18:
 if s1.ObjectName <> s2.ObjectName:
change it to:
 if s1.ObjectName != s2.ObjectName:

At least this makes the workbench load properly again. Please check, if it makes sense and apply it as soon as possible. 
I already learned into FC for some weeks and I like it and I'd find it disappointig if this worthy assembly feature addon -- not working -- would hinder testers from furtherly using FC.

Best regards,
Manuel Krause

normandc, 2018-04-01:
Manuel, this repository is no longer maintained. hamish2014 abandoned it, and r-frank (BPLRFE) who took charge of it died last September.

ManuelKrause, 2018-04-01:
Yes, it's sad. I just read about it some minutes ago, before your info.
And now?

normandc, 2018-04-01:
And now, what can I tell you? Your fix proposal will remain unanswered unless someone contacts hamish2014 to claim maintainer-ship of this repo.

ManuelKrause, 2018-04-01:
Thank you for your quick replies. 
Can't the master's master chief pick the fix in?

For now, I'm too far away from python programming knowledge, to even think of an application. ;-)

normandc, 2018-04-01:
Do you actually read what I am saying? This repository is owned by hamish2014. It is separate from the FreeCAD project. hamish2014 abandoned assembly2 development. BPLRFE took responsability for it but died. There is *no* "master's master chief" left!!!

ManuelKrause, 2018-04-01:
I didn't want to sound impolite at all.
Maybe at best, find someone to take it over.

ManuelKrause, 2018-04-01:
Sorry, even my last message can be misunderstood.
I meant a question: Does BPLRFE's death endanger the whole FreeCAD project?
Or only assembly2?

normandc, 2018-04-01:
I already wrote that assembly2 is separate from the FreeCAD project. Just go look at the FreeCAD's github repo and see for yourself if it seems endangered. https://github.com/FreeCAD/FreeCAD

ManuelKrause, 2018-04-01:
O.k. It's alive!
Maybe my fix proposal helps FC 0.17 users to use the assembly2 addon.

Best regards, Manuel Krause

ManuelKrause, 2018-04-07:
ATM I feel quite sorry for having made this noise one here. I was quite naive and highly uninformed about some facts and "behind-the-scenes" things of FreeCAD.

Last week I realised, that only fixing that very file in my first posting, doesn't help at all to make it work. During the last days I digged into it further, and changed several files of "assembly2" on here, so that I really got a glance of what it's capable of.  I'm no programmer, so the changes are based on trial-and-error and searches upon existing code, locally and via github.

Today I've even been able to make a crankshaft out of many many parts (v0.17).

Many error messages remain to be debugged, but the first result is nice.

Best regards, Manuel Krause

kbwbe, 2018-07-15:
Hi Manuel,
as your are already working with A2plus and we did a lot of communication, i close this issue.

closed 2018-07-15 by kbwbe