hamish2014/FreeCAD_assembly2 #156: What is the status of this module today?

abetis, 2018-07-12:
Hello gents,

From reading other posts I understand that @hamish2014 abandoned the project.
@BPLRFE took the maintenance, but passed away on 09/2017.

So what's the future of that work?
Assembly module is **_THE_** module I need to use FreeCAD.

I see some commits by @hamish2014 , @MrGradgrind and @kbwbe few days ago, does that mean there is someone who continue the development or some kind of maintenance?

Is there an alternative to assembly2 or a maintained fork of that project?

I'm a programmer and willing to help, but I'm not sure what version is maintained, if at all...

kbwbe, 2018-07-15:
Hello abetis,
hamish2014 allowed me to do push's to his repository. So there is the possibility again to change something within. I do not intend to do big changes on this workbench, only some absolute necessary things to keep it running for people who are still using it and are having old assembly files they need. There is a new open branch "updates-2018-07". You can check out and test/improve + do a pullrequest. But please in a small limited way.

I am working on my own new workbench (working with similar concept) which you can find here:
[https://github.com/kbwbe/A2plus](url)
You also can search in FreeCAD forum for it. This new WB is under active development and i concentrate all work on this new one.

closed 2018-07-15 by kbwbe

abetis, 2018-07-15:
What about the Assembly3 work done by realthunder?
https://forum.freecadweb.org/viewtopic.php?f=20&t=25712
It looks very promising.

Why don't you join the forces?